ros2/launch #750: Requirement:Only start all nodes and then this ros2 launch process exit

lianggangMei, 2023-12-19:
When we launch al lot of nodes with run/run_async in ros2 launch, because ros2 launch process has high cpu uasge so we don't need an event loop and visit all entities of all included LaunchDescription instances.  Only start all nodes and then this ros2 launch process exit, which is similar to:
ros2 run node1
ros2 run node2
ros2 run node3
ros2 run node4
...
exit

     def run(self, *, shutdown_when_idle=True) -> int: 
         """ 
         Run an event loop and visit all entities of all included LaunchDescription instances. 
  
         This should only ever be run from the main thread and not concurrently with 
         asynchronous runs (see `run_async()` documentation).

clalancette, 2023-12-20:
I guess I'm not sure why you would use `launch` at all in this situation.  Couldn't you just write a bash script that did this instead?  What would be the advantage of having this in `launch`?

lianggangMei, 2023-12-21:
Thanks clalancette
because we need LaunchDescription py as below to config, it is easier/better to maintain and extend for our projects.

from launch import LaunchDescription
from launch_ros.actions import Node

# here we can do a lot of things.
...

def generate_launch_description():
    return LaunchDescription(
        [
            Node(

closed 2023-12-28 by lianggangMei