ros2/launch #765: pytest 8.1 incompatibility: TypeError: import_path() missing 1 required keyword-only argument: 'consider_namespace_packages'

jtbandes, 2024-03-12:
## Bug report

**Required Info:**

- Operating System:
  - GitHub Actions ubuntu-22.04
- Installation type:
  - binaries via action-ros-ci

```
      platform linux -- Python 3.10.12, pytest-8.1.1, pluggy-1.4.0
      cachedir: build/foxglove_msgs/ament_cmake_pytest/foxglove_msgs_test_msgs/.cache
      rootdir: /__w/schemas/schemas/ros_ws
      plugins: launch-testing-ros-0.26.4, launch-testing-3.4.0, ament-pep257-0.16.3, ament-flake8-0.16.3, ament-xmllint-0.16.3, ament-lint-0.16.3, ament-copyright-0.16.3, mock-3.12.0, cov-4.1.0, rerunfailures-13.0, colcon-core-0.15.1, repeat-0.9.3
```

#### Steps to reproduce issue
A build failure was encountered here (for Humble, Iron, and Rolling): https://github.com/foxglove/schemas/actions/runs/8241439601/job/22538701279

```
  /usr/bin/bash -c source /opt/ros/rolling/setup.sh && colcon test-result --verbose
  /usr/local/lib/python3.10/dist-packages/colcon_core/argument_parser/type_collector.py:9: UserWarning: 'colcon_core.argument_parser.type_collector.TypeCollectorDecorator' has been deprecated
    class TypeCollectorDecorator(ArgumentParserDecorator):
  /usr/local/lib/python3.10/dist-packages/colcon_core/argument_parser/type_collector.py:62: UserWarning: 'colcon_core.argument_parser.type_collector.SuppressTypeConversions' has been deprecated, use 'colcon_core.argument_parser.action_collector.SuppressTypeConversions' instead
    class SuppressTypeConversions:
  build/foxglove_msgs/Testing/20240312-0011/Test.xml: 34 tests, 0 errors, 1 failure, 0 skipped
  - foxglove_msgs_test_msgs
    <<< failure message
      -- run_test.py: invoking following command in '/__w/schemas/schemas/ros_ws/build/foxglove_msgs':
       - /usr/bin/python3.10 -u -m pytest /__w/schemas/schemas/ros_ws/src/l3sikm3bb4/schemas/ros_foxglove_msgs/test/test_msgs.py -o cache_dir=/__w/schemas/schemas/ros_ws/build/foxglove_msgs/ament_cmake_pytest/foxglove_msgs_test_msgs/.cache --junit-xml=/__w/schemas/schemas/ros_ws/build/foxglove_msgs/test_results/foxglove_msgs/foxglove_msgs_test_msgs.xunit.xml --junit-prefix=foxglove_msgs
      ============================= test session starts ==============================
      platform linux -- Python 3.10.12, pytest-8.1.1, pluggy-1.4.0
      cachedir: build/foxglove_msgs/ament_cmake_pytest/foxglove_msgs_test_msgs/.cache
      rootdir: /__w/schemas/schemas/ros_ws
      plugins: launch-testing-ros-0.26.4, launch-testing-3.4.0, ament-pep257-0.16.3, ament-flake8-0.16.3, ament-xmllint-0.16.3, ament-lint-0.16.3, ament-copyright-0.16.3, mock-3.12.0, cov-4.1.0, rerunfailures-13.0, colcon-core-0.15.1, repeat-0.9.3
      collected 0 items / 1 error
      
      ==================================== ERRORS ====================================
      ________ ERROR collecting src/l3sikm3bb4/schemas/ros_foxglove_msgs/test ________
      /usr/local/lib/python3.10/dist-packages/pluggy/_hooks.py:501: in __call__
          return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
      /usr/local/lib/python3.10/dist-packages/pluggy/_manager.py:119: in _hookexec
          return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
      /usr/local/lib/python3.10/dist-packages/_pytest/python.py:225: in pytest_collect_file
          module: Module = ihook.pytest_pycollect_makemodule(
      /usr/local/lib/python3.10/dist-packages/_pytest/config/compat.py:79: in fixed_hook
          return hook(**kw)
      /usr/local/lib/python3.10/dist-packages/pluggy/_hooks.py:501: in __call__
          return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
      /usr/local/lib/python3.10/dist-packages/pluggy/_manager.py:119: in _hookexec
          return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
      /opt/ros/rolling/lib/python3.10/site-packages/launch_testing/pytest/hooks.py:188: in pytest_pycollect_makemodule
          entrypoint = find_launch_test_entrypoint(path)
      /opt/ros/rolling/lib/python3.10/site-packages/launch_testing/pytest/hooks.py:178: in find_launch_test_entrypoint
          module = import_path(path, root=None)
      E   TypeError: import_path() missing 1 required keyword-only argument: 'consider_namespace_packages'
      - generated xml file: /__w/schemas/schemas/ros_ws/build/foxglove_msgs/test_results/foxglove_msgs/foxglove_msgs_test_msgs.xunit.xml -
      =========================== short test summary info ============================
      ERROR ../../src/l3sikm3bb4/schemas/ros_foxglove_msgs/test - TypeError: import_path() missing 1 required keyword-only argument: 'consider_namespace_packages'
      =============================== 1 error in 0.10s ===============================
      ERROR: found no collectors for /__w/schemas/schemas/ros_ws/src/l3sikm3bb4/schemas/ros_foxglove_msgs/test/test_msgs.py
      
      -- run_test.py: return code 4
      -- run_test.py: verify result file '/__w/schemas/schemas/ros_ws/build/foxglove_msgs/test_results/foxglove_msgs/foxglove_msgs_test_msgs.xunit.xml'
    >>>
  build/foxglove_msgs/test_results/foxglove_msgs/foxglove_msgs_test_msgs.xunit.xml: 1 test, 1 error, 0 failures, 0 skipped
  - foxglove_msgs src.l3sikm3bb4.schemas.ros_foxglove_msgs.test
    <<< error message
      collection failure
    >>>
  
  Summary: 4190 tests, 1 error, 1 failure, 1363 skipped
```

#### Additional information
This `consider_namespace_packages` arg appears to have become required in pytest 8.1 (https://github.com/pytest-dev/pytest/pull/11997). This is also causing issues for other pytest plugin authors (https://github.com/scientific-python/pytest-doctestplus/issues/240). You can see an example fix at https://github.com/scientific-python/pytest-doctestplus/pull/241.

xydesa, 2024-03-12:
My team is seeing this issue as well.

tonynajjar, 2024-03-12:
We're also facing this issue. It's say to safe it's a bug affecting everyone

tonynajjar, 2024-03-12:
Created PR https://github.com/ros2/launch/pull/766

ali-bdai, 2024-03-13:
@dyershov-bdai
I guess this is what we're running into as well. Thanks @tonynajjar! for the fix :)

closed 2024-03-13 by clalancette