ros2/launch #780: Return Code not set on SIGINT

DLu, 2024-05-16:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```python
from launch import LaunchDescription
from launch.actions import ExecuteProcess

def generate_launch_description():
    ld = LaunchDescription()
    ld.add_action(ExecuteProcess(cmd=['sleep', '5']))
    return ld
```

#### Expected behavior
 * If you run the command `sleep 5` normally, it has return code 0. 
 * If you run it and hit CtrlC, it has the return code 130. 
 * If you launch the above lunch file normally, it has the return code 0. 
 * If you run it and hit CtrlC, I expect it to return some non-zero code. 

#### Actual behavior
In actuality, it also returns 0. 

#### Additional information
The return code for launch [is set on exceptions](https://github.com/ros2/launch/blob/eea42eb2948a7ba9d893b50a9dc7a0006d4be632/launch/launch/launch_service.py#L281-L285) but [the sigint handler](https://github.com/ros2/launch/blob/eea42eb2948a7ba9d893b50a9dc7a0006d4be632/launch/launch/launch_service.py#L191-L202) does not set the return code and shutsdown the processes "normally". 

The best workaround I came up with is an `OnShutdown` event handler with an `OpaqueFunction` to save the value of `event.reason` but that doesn't seem great.

audrow, 2024-05-23:
I'm going to mark this as backlog.

It's not 100% clear to me what the right behavior is here. What did we do in ROS 1? There also may be some funniness with Windows.

mjcarroll, 2024-05-23:
So the longer discussion was that it should probably return 130 on ctrl-c, since that is the error code that represents that an interactive application was interrupted.

I don't think anybody would be opposed to `launch` having that behavior as well.

Adjacent to that, it would be interesting to know how `roslaunch` did it, and if not returning `130` was keeping in line with that precedent or just an oversight.

Additionally, all bets are off with Windows and how it handles interrupts and return codes, so it would be somewhat worthwhile to try to make the behaviors consistent.

DLu, 2024-05-23:
Sad to report that `roslaunch` returns 0 in most cases except for [exceptions](https://github.com/ros/ros_comm/blob/845f74602c7464e08ef5ac6fd9e26c97d0fe42c9/tools/roslaunch/src/roslaunch/__init__.py#L359) (which sigint does not generate)