ros2/launch #800: Cant build launch_testing_ament_cmake

wscott20, 2024-10-09:
Every time I try to colcon build launch_teting_ament_cmake it keeps on saying
```--- stderr: launch_testing_ament_cmake                                          
CMake Error at CMakeLists.txt:24 (message):
  launch_testing package not found

---
Failed   <<< launch_testing_ament_cmake [11.6s, exited with code 1]
```
No matter what I try it won't build it even after colcon builds and installs launch_testing
## Bug report

**Required Info:**

- Operating System:
  - MacOS Sequoia
- Installation type:
  - source
- Version or commit hash:
  - `df54d3b755e3b5fc73972f4e97b0b18a1bbc61db`
- DDS implementation:
  - fastrtps/fastdds
- Client library (if applicable):
  - (trying to use rclpy but it won't build since it needs rcutils which needs launch_testing_ament_cmake)

#### Steps to reproduce issue
```
mkdir -p ros2_ws/src
cd ros2_ws
vcs import --input https://raw.githubusercontent.com/ros2/ros2/jazzy/ros2.repos src
(maybe remove unnecessary packages that arentneeded to build for rclpy and ros2cli)
colcon build
#### Expected behavior
being able to build everything and source install/setup.bash and run ros2 commands
#### Actual behavior
```
Starting >>> launch_testing_ament_cmake
--- stderr: launch_testing_ament_cmake                         
CMake Error at CMakeLists.txt:24 (message):
  launch_testing package not found

---
Failed   <<< launch_testing_ament_cmake [2.57s, exited with code 1]
```
#### Additional information
tried on humble and jazzy

cottsay, 2024-11-01:
Hi @wscott20, was this resolved by straightening out the Python/Conda issues we discussed in Discord, or is this still a problem for you?

wscott20, 2024-11-01:
I got it working via docker

cottsay, 2024-11-01:
I see, thanks for the response.

Without more information about the environment in which you're seeing this issue, I'm not able to reproduce the error you're seeing. If you decide you'd like to continue debugging this issue, please feel free to re-open the ticket and provide more information about your system that might help us pinpoint what's triggering the misbehavior.

Thanks!

closed 2024-11-01 by cottsay

mikeslembrouck, 2025-04-02:
Hi,

I seem to have the exact same error while trying to build ROS2 from source on MacOS Sequoia 15.3. I'm using the same steps as the initial bug report.

Here's a snippet of the error message:

`......
Finished <<< launch_testing [3.99s]
Starting >>> launch_testing_ament_cmake
......
--- stderr: launch_testing_ament_cmake

CMake Error at CMakeLists.txt:24 (message):
  launch_testing package not found

Failed   <<< launch_testing_ament_cmake [2.51s, exited with code 1]`

Here are some specs of my build environment:

- Ros2 launch repo commit-id: 37431e64cf8f0c69ce7271ce539c5e48bbf38ac4
- Apple clang version 17.0.0
- MacOS SDK 15.3
- Python 3.13.2
- cmake version 3.31.6

Any other information I can provide you with that can help to tackle the build issue?

Thanks in advance!

wscott20, 2025-04-02:
> Hi,
> 
> I seem to have the exact same error while trying to build ROS2 from source on MacOS Sequoia 15.3. I'm using the same steps as the initial bug report.
> 
> Here's a snippet of the error message:
> 
> `...... Finished <<< launch_testing [3.99s] Starting >>> launch_testing_ament_cmake ...... --- stderr: launch_testing_ament_cmake
> 
> CMake Error at CMakeLists.txt:24 (message): launch_testing package not found
> 
> Failed <<< launch_testing_ament_cmake [2.51s, exited with code 1]`
> 
> Here are some specs of my build environment:
> 
> * Ros2 launch repo commit-id: [37431e6](https://github.com/ros2/launch/commit/37431e64cf8f0c69ce7271ce539c5e48bbf38ac4)
> * Apple clang version 17.0.0
> * MacOS SDK 15.3
> * Python 3.13.2
> * cmake version 3.31.6
> 
> Any other information I can provide you with that can help to tackle the build issue?
> 
> Thanks in advance!

I’d recommend trying docker there should be a tutorial on it in the docs