ros2/launch #801: Allow optional scoping for including a launchfile

SuperJappie08, 2024-10-09:
## Feature request
`IncludeLaunchDescription` is not scoped if I understand correctly. (Based on #313 #689 and [the design docs](https://design.ros2.org/articles/roslaunch.html)

#### Feature description

An extra argument for `IncludeLaunchDescription` action, which could enable scoping.
This would prevent the leakage of arguments.

#### Implementation considerations

This is already possible by including into a group with `forwarding` set to false.
Default could be false to not break the current behavior.

wjwwood, 2025-02-28:
@SuperJappie08 thanks for the contribution, it's definitely something that others have run into and found surprising.

I really like the idea of a new Action, like `ScopedIncludeLaunchDescription` or similar, and I would be in favor of including it in this repository if you want.

Also, I really appreciate you making it into a stand-alone package as a first step. It definitely helps demonstrate the idea and value, and it makes it easier for folks to try it or use it, even in older releases of ROS 2.

alsora, 2025-03-13:
Hi @SuperJappie08 any update on this?

SuperJappie08, 2025-03-17:
> Hi [@SuperJappie08](https://github.com/SuperJappie08) any update on this?

Hi,

I was initially waiting on a response from a maintainer before spending a lot of time on this because I have a tendency to spend a lot of time on the wrong things.
However, some family stuff has come up recently, which required a shuffle in my priorities, so I have not had the time (yet) to start this.

I think I would be able to start this week.

Timple, 2025-03-17:
>  it's definitely something that others have run into and found surprising.

Certainly!

So why not make this the default? Breaking changes are allowed between ROS releases. 

I know breaking backward compatibility should not be considered lightly. But it's a trade-off between:
- new developers running into this surprising feature
- existing developers actually using this feature

The first group might actually be larger than the second.

SuperJappie08, 2025-03-17:
> Certainly!
>
> So why not make this the default? Breaking changes are allowed between ROS releases.
>
> I know breaking backward compatibility should not be considered lightly. But it's a trade-off [...]

Having it as a separate action, could mean the default include action could change.
But I think your point is valid, and it possibly could make sense to rename the old `IncludeLaunchDescription` to `IncludeUnscopedLaunchDescription` so the new scoped version could be the easier/shorted(/ more likely to be adopted as the default) name (`IncludeLaunchDescription`).

Normally, I would also cheer for backwards compatibility, but since the unscoped version feels counterintuitive, it might make sense to _fix_ it.

SuperJappie08, 2025-03-18:
I found a fatal flaw after doing more testing on this feature with an implementation similar to my initial attempt in the package.

Currently, the 'scoped' is essentially a normal include wrapped in a scoped non-forwarding group, which only keeps the provided launch arguments.
This gives essentially the expected behavior from this feature, however, since **only** the launch argument to the nested files are kept in systems depending on non-user set launch configurations, such as the `PushRosNameSpace` and `SetRemap` actions from `launch_ros`.

My initial reaction was to attempt to include launch configurations which were not defined explicitly by the child file. 
However, this requires a lot of rework to support multi-nesting situations.

Since not being able to do things, like name spacing a launch file, would not resolve the current non-scoping confusion behavior, I am unsure on how to continue (for now).

Does anyone have suggestions for a direction to move forward? (I will try to think of reasonable solutions as well)

SuperJappie08, 2025-03-18:
Another option could be to register overscoping launch configurations in the context, which will always be included.
This could be implemented as a launch configuration, which contains the keys of the unscopable launch configurations.

SteveMacenski, 2025-04-29:
I'd love to have something like this in the package (and by default!) :-)

SuperJappie08, 2025-05-01:
@alsora, an initial working implementation is complete (#841).

ijnek, 2025-06-17:
Sorry — I completely missed your comment (https://github.com/ros2/launch/pull/841#issuecomment-2845561097) mentioning that you had already worked around this issue using global launch configurations.

Personally, I find it a bit odd that we use launch configurations to pass ros_remaps, ros_namespace, and global_params to included launch files. While looking into launch, it seems we could instead store remapping and namespacing info in the local stack rather than using launch configurations.

https://github.com/ros2/launch/blob/820e5a2c57dafd807ea4cdf33952e87c78929ce5/launch/launch/launch_context.py#L63-L64

I’ve drafted a PR to explore this idea: https://github.com/ros2/launch_ros/pull/470.

This approach would avoid the need to track a global list of which launch configurations are meant to propagate. Let me know what you think.

SuperJappie08, 2025-06-17:
That would indeed also resolve those weird behaviors.

Personally, I'm not set to on the global exclusion list, since it is unclear in the internal API, since there are also global variables in the launch context. So naming and behavior of the global list is not very transparent in my opinion.

Currently, I'm mostly curious, why these `launch_ros` features use the launch configurations instead of launch context locals and why that decision was made in the first place.
Because if it was a conscious decision to do it like this, there might be a valid reason we are currently missing.

SuperJappie08, 2026-03-19:
I think the main difference with this original attempt compared to #953 is that my attempt also blocks not explicitly passed launch configurations, causing the namespacing issue.
However, this extra blocking prevents the included launch files from being affected by the parent (which can be useful, for example, for external modules.).

> Plus autoware_launch is _deliberately_ using scope leakage in their recent commits which haven't passed a day from this reply. (This is also mentioned in the description of [#953](https://github.com/ros2/launch/pull/953) .)

To my understanding, this behavior of _leaking_ is actually intended behavior, as normal `include` is a flat copy-past style action.
(Which can be really useful from time to time.)

paulsohn, 2026-03-19:
> To my understanding, this behavior of *leaking* is actually intended behavior, as normal `include` is a flat copy-past style action.

Agreed, assuming that if the user is always sure what they're doing. Autoware is suffering from unintended leaks.

I think it is sufficient and the clearest for having:
* `<include scoped="true/false" (...) />` should directly correspond to `<group scoped="true/false"><include (...) /></group>`
* `<include scoped="true" forwarding="true/false" (...) />` should directly correspond to `<group scoped="true" forwarding="true/false"><include (...) /></group>`

and I feel we better not have a design that `<include scoped="true" (...) />` sets up both scoping and blocking forwarding. 
If we want it, we can add `forwarding` attribute separately on top of #953 .

wjwwood, 2026-08-18:
Sorry for the late reply, I've been searching for time to catch up on the conversation and weigh in. 

I'm gonna tl;dr my opinions here and then follow with details for my rationale:

- `IncludeLaunchDescription` being unscoped was intentional
  - I find it helpful to consider launch files as markup and not code, and so it was intentionally made to resemble systems that modify text, not the semantics of code
  - it resembles C++'s preprocessor `#include` statement, i.e. an include statement includes the content in-place as if it were literally written there, but not the C++ compiler which considers the semantics of the content
  - also resembles most templating engine's include statements (most are not scoped in my experience)
  - as opposed to CMake's include system where you must explicitly use PARENT_SCOPE to write to the parent scope
  - this behavior makes sense (in my opinion) when breaking up a launch file for a single task into multiple files, or when reusing fragments in multiple places
  - this is surprising behavior (in my opinion) when making top-level launch files that include unrelated launch files (each specific to an entire task) together
    - the idea was that top-level launch files would use scoped group actions to resolve this
- `IncludeLaunchDescription` including the parent scope and also taking `launch_arguments` was intentional
  - again this behavior (including the parent scope by default) makes sense if you consider it to be a parallel to C++'s `#include` statement or a template engine's include
  - the purpose of `launch_arguments` was to explicitly check that required launch arguments in the included launch file were provided, and to produce earlier feedback when you failed to provide them
  - you can do basically everything without `launch_arguments` since the parent scope is there, but speculative error checking was difficult without it, however I think there's an argument to made that this argument was a mistake, as it mixes content manipulation (including the sub-launch description) and semantics (were the require arguments of the sub-launch description satisfied)
  - `launch_arguments` also help you to be explicit when remapping arguments from the parent scope, but that could also be done with `SetLaunchConfiguration`, e.g. when the parent launch file is multiplexing options to sub launch files like `<include ...><let name="name" value="$(arg left_sensor_name)"/></include>` could also just be `<group><let name="name" value="$(arg left_sensor_name)"/><include ...></include></group>`
  - including a launch file but not providing the parent scope is a rarer use case (in my opinion), and I personally don't find it surprising that this is the default behavior
- The `Group` action's purpose is to provide scoping and forwarding control for a group of actions, duplicating this behavior in the `IncludeLaunchDescription` action causes unnecessary overlap between the classes
  - what happens when someone updates the behavior for the group action and forgets to do so for the include action? or vice versa? keeping scoping and forwarding in the group action is good for code cohesion (in my opinion)
  - why not drop group and just have scoping and forwarding on include? it is useful to be able to control scoping and forwarding on a group of actions without putting them in a separate launch file
- We can (and should) improve the ergonomics of these workflows and reduce surprise where possible
  - I would prefer to avoid arguments that change the behavior of `IncludeLaunchDescription`, especially ones that overlap with `Group`
  - instead limit changes to the frontend, e.g. `<include scoped=true ...>...</include>` in XML is ok with me for syntactic sugar, but under the hood it would make more sense as a `Group` action that contains an `IncludeLaunchDescription` action
  - if we must have a new Python class, name it differently (use the type system to convey behavior changes, not arguments), e.g. `ScopedIncludeLaunchDescription`, but that class should explicitly be a wrapper for a `Group` action that contains a `IncludeLaunchDescription` instance and provides the frontend syntactic sugar

I think I'll stop there, since this post is already long, but I'm happy to continue the conversation on any of these particular points.

Also, I usually don't have time to join the ROSGraph working group, but if you'd like me to join a specific instance of it to discuss this or even have a dedicated meeting to discuss these issues, let me know. I'm happy to try and arrange that.

peci1, 2026-08-18:
> * instead limit changes to the frontend, e.g. `<include scoped=true ...>...</include>` in XML is ok with me for syntactic sugar, but under the hood it would make more sense as a `Group` action that contains an `IncludeLaunchDescription` action

I fully agree with this kind of solution.