ros2/launch #803: Action for runtime modifiable parameters in YAML config

mhubii, 2024-10-22:
Hi there, not very certain where to raise this issue. Please redirect in case.

To my best understanding, parameters, as specified in e.g. `yaml`, are fixed

```yaml
my_node:
  ros__parameters:
    key: value
```

i.e. `value` cannot be changed easily. 

There are suggestions to parse the `yaml` file inside launch, then overwrite. However, this seems quite cumbersome yet a frequent task.

I don't understand the inner workings well enough, but there exist libraries that allow for runtime parameter modification / addition (https://hydra.cc/docs/intro/).

I see that dynamically adding parameters might not work in C++, but is there an `Action` that would allow modification during launch?

```python
yaml_config = "path_to_config"
modified_yaml_parameters = YAMLModify(yaml_config, {"my_node.key = value")
```

fujitatomoya, 2024-10-25:
@mhubii 

correct me if i am mistaken, but your requirement is to change the parameter (default) values loaded from parameter `yaml` during launch? the use case is that we have default parameter settings in the yaml parameter file, but some values need to be adjusted to the nodes, is that correct?

how about using `DeclareLaunchArgument` and add it as `LaunchConfiguration` to `parameters` in `Node`? so that parameters are loaded from yaml file and overwrite the parameters with `LaunchConfiguration`?

mhubii, 2024-10-29:
closing this issue because the immediate problem can be solved through overwriting, and anything else seems beyond scope.

closed 2024-10-29 by mhubii