ros2/launch #815: argument collision in included files (xml and yaml)

asherikov, 2024-12-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - humble
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

test.yaml
```
   launch:
       - include:
             file: "$(find-pkg-share my_package)/launch/inc1.yaml"
       - include: 
             file: "$(find-pkg-share my_package)/launch/inc2.yaml"
```
inc1.yaml
```
   launch:
       - arg: {name: "my_parameter", default: "my_parameter_value_1"}
       - executable:
             cmd: "echo inc1 $(var my_parameter)"
             output: "screen"
```
inc2.yaml
```
   launch:
       - arg: {name: "my_parameter", default: "my_parameter_value_2"}
       - executable:
             cmd: "echo inc2 $(var my_parameter)"
             output: "screen"
```

Output:
```
   [INFO] [echo-1]: process started with pid [86076]
   [INFO] [echo-1]: process has finished cleanly [pid 86076]
   [INFO] [echo-2]: process started with pid [86078]
   [echo-1] inc1 my_parameter_value_1
   [INFO] [echo-2]: process has finished cleanly [pid 86078]
   [echo-2] inc2 my_parameter_value_1
```

#### Actual behavior
Default value of `my_parameter` in `inc2` is overrided by default value of parameter with the same name in `inc1`

#### Expected behavior
Parameters in separate files should not collide.

#### Additional information

The issue can be avoided by additionally scoping each include with `group`, however, I strongly believe that scoping should be implied by inclusion.

asherikov, 2025-02-06:
hi,

1. Yes, duplicated variables need to be handled in a better way even if lack of scoping is intentional. Unfortunately multiple declarations cannot be an error in this case: parameter has to be declared in each launch file if they might be used separately. I dont see a clean workaround for this design flaw that would prevent bugs due to collisions without breaking compatibility. Printing a warning however is a must.

2. Sorry, I havent tried to use python scripts, may be later.

wjwwood, 2025-02-28:
For those wanting a scoped include, @ implemented that in his own package here:

https://github.com/ros2/launch/issues/801

We could consider integrating that into this repository, either as an opt-in option on `IncludeLaunchDescription` (and related XML/YAML invocation) or as a new action. But I do not think we can consider changing the existing behavior.

SteveMacenski, 2025-04-29:
I just found this thread after filing with https://github.com/ros2/launch_ros/issues/462 -- its (probably obviously) also happening with Python3, not just XML and YAML. 

@wjwwood I think having that scopred include in this repository would be helpful indeed from https://github.com/ros2/launch/issues/801. I think this should probably be the default behavior in my opinion