ros2/launch #821: Add logging actions for severity levels other than 'INFO'

AntoSave, 2025-01-28:
Currently there is no action analogous to `LogInfo` for other severity levels such as 'WARN', 'DEBUG' and 'ERROR'.
I was wondering whether this is a design choice or a feature that is yet to be implemented.

Yadunund, 2025-02-06:
@AntoSave thanks for the ticket. The feature you propose is worthwhile to implement and we would be happy to review a PR for it if you or anyone else decides to take it up. Please let me know if this is something you're willing to work on.

One consideration to be mindful of is whether having an `Error` log level makes sense for launch.

christophebedard, 2025-02-06:
Maybe a new action could be added to log a message with a given logging level, e.g., `level='info'`, `level='warning'`, etc. Then [`LogInfo`](https://github.com/ros2/launch/blob/a9258c3e23238be022869edca19c67d45a932d6e/launch/launch/actions/log_info.py#L32) could just be a simple wrapper around that (with `level='info'`).

PR #858 (2025-04-28): Add Other Logging Implementations
Added generic `Log` class and Specializations for `LogInfo`, `LogDebug`, `LogError`, `LogWarning`.

Closes #821