ros2/launch #826: Add OpaqueFunction to list at docs.ros.org

EmmanuelMess, 2025-02-07:
Searching for OpaqueFunction in docs.ros.org/en/rolling/p/launch I find nothing, it should be added: https://docs.ros.org/en/rolling/p/launch/search.html?q=OpaqueFunction&check_keywords=yes&area=default

christophebedard, 2025-02-20:
I think this might be another case of the documentation generation (rosdoc2) breaking for some reason.

A search of "OpaqueFunction" doesn't return anything because the class itself isn't included in the docs. However, the module itself (`opaque_function`) is included. It should include the `OpaqueFunction` class, but it doesn't: https://docs.ros.org/en/rolling/p/launch/launch.actions.opaque_function.html.

For example, this page for the `launch_ros.actions.node` module doesn't show anything in the Humble docs: https://docs.ros.org/en/humble/p/launch_ros/launch_ros.actions.node.html. It was fixed in https://github.com/ros2/launch_ros/pull/425, so in Rolling the docs look for that module look much better: https://docs.ros.org/en/rolling/p/launch_ros/launch_ros.actions.node.html.

If I run `rosdoc2` on `launch` (@ Rolling) locally, I get this:

[image]

Not sure why it's not showing up on https://docs.ros.org/en/rolling/p/launch/launch.actions.opaque_function.html. Any help would be appreciated.

EmmanuelMess, 2025-02-20:
> Not sure why it's not showing up on [docs.ros.org/en/rolling/p/launch/launch.actions.opaque_function.html](https://docs.ros.org/en/rolling/p/launch/launch.actions.opaque_function.html). Any help would be appreciated.

If you add a link to your last CI execution, I can try to find a fix.

gaolebaigao, 2026-08-27:
Hi, I'd like to investigate this issue.

My plan is to reproduce the missing `OpaqueFunction` API documentation using the current Rolling branch and `rosdoc2`, then compare its generated Sphinx output with another documented launch action.

I will first determine whether the root cause is:

1. package/module metadata or documentation configuration in `ros2/launch`, or
2. API discovery behavior in `rosdoc2`.

If it is launch-side, I will prepare a minimal documentation/configuration fix with a reproducible documentation build check. If it is rosdoc2-side, I will provide a minimal reproducer and move the fix to the appropriate repository.

Does this scope sound appropriate before I start implementing it?