ros2/launch #830: Bug or Feature? Subprocess receive CTRL+C sent to launch process

huweiATgithub, 2025-02-24:
# Step to Reproduce

## Create launch files
The launch file:

```python
import time
from pathlib import Path

from launch import LaunchDescription
from launch.actions import ExecuteProcess, RegisterEventHandler
from launch.event_handlers import OnShutdown

def on_shutdown(_, __):
    print(f"Shutdown event received at {time.time()}")
    time.sleep(5)
    print(f"Shutting event handled at {time.time()}")

def generate_launch_description():
    # Get path to our test script
    signal_test_path = Path(__file__).parent / "signal_test.py"
    # Launch our test process
    test_process = ExecuteProcess(
        cmd=["python3", str(signal_test_path.absolute())],
        output="screen",
        name="signal_test",
        additional_env={"PYTHONUNBUFFERED": "1"},
    )
    register_handlers = RegisterEventHandler(
        event_handler=OnShutdown(on_shutdown=on_shutdown),
    )

    return LaunchDescription([test_process, register_handlers])
```

In the same folder, create the process file, `signal_test.py`:

```
#!/usr/bin/env python3
import signal
import sys
import time

def signal_handler(sig, frame):
    print(f"\nSIGINT (Ctrl+C) received at {time.time()}")
    print("Cleaning up...")
    time.sleep(1)  # Simulate cleanup
    print("Exiting gracefully")
    sys.exit(0)

def main():
    signal.signal(signal.SIGINT, signal_handler)
    print("Starting signal test program...")
    print("Press Ctrl+C to exit")

    counter = 0
    while True:
        print(f"Running... count: {counter}")
        time.sleep(1)
        counter += 1

if __name__ == "__main__":
    main()
```

## Run

```
ros2 launch <the-launch-file>
...
CTRL+C
```

The output is similar to

```
Shutdown event received at 1740388278.8298662
Shutting event handled at 1740388283.830014
[signal_test-1] SIGINT (Ctrl+C) received at 1740388278.8293986
```

Note that the output in the subprocess `signal_test-1` has a timestamp earlier than the Shutdown handler handled the Shutdown signal.

# Summary
This implies that the `CTRL+C` signal is sent to the subprocess directly instead of being managed by the LaunchService.

Is this by design? I think it is more reasonable to let the launch service manage signals sent to subprocesses. For example, one might need more granularity during shutdown.

huweiATgithub, 2025-03-03:
Any insight for the issue?

wjwwood, 2025-03-13:
This is by design, as it is what happened with ROS 1's launch system and it's similar to what happens if you have child processes in a shell script or a python script using `subprocess`. Signals are propagated by default.

We also have some behavior where if the child process does not exit quickly launch will escalate the signals it sends to the processes to SIGTERM and then SIGKILL after a configurable about of time. Those behaviors are configurable and optional, but not sending ctrl-c (SIGINT) isn't an option atm as far as I can remember, but expanded launch's API to allow for this kind of customization would be fine, as long as it's optional.

I'm going to close this for now, but feel free to continue discussion here or to open a new feature request issue detailing what you think should be possible and/or what should change in launch.

closed 2025-03-13 by wjwwood