ros2/launch #832: conditional tag does not work

pitosalas, 2025-03-03:
I have tried this and many variations. 

```
  includes:
    - file: $(find-pkg-share my_package)/launch/optional_components.launch.yaml
      condition:
        value: $(var use_optional_component)
        type: 'boolean'
```

I have trolled the source code (e.g. ros2/launch_yaml) and can't figure out where the relevant code is. But I would be happy to offer support for condition if it is not already provided.

christophebedard, 2025-03-03:
Could you provide more information? For example:

* How you're launching the launch file
* What you expect to happen, what actually happens
* The full launch file + the full included launch file (`optional_components.launch.yaml`), or a minimal subset of it that allows someone to reproduce the issue
* Which ROS 2 distro you're using
* Any other relevant information

Some info that might help you when looking at the source code:

* The code for the `IncludeLaunchDescription` action is here: https://github.com/ros2/launch/blob/bdf124d83c847b5205c204932e9f86daccc0c747/launch/launch/actions/include_launch_description.py
    * Note that the name of the action for frontends is `include` and not `includes` (see the `@expose_action()` decorator)
    * You can find some examples here: https://docs.ros.org/en/rolling/How-To-Guides/Launch-file-different-formats.html#launch-file-examples
* The condition feature is part of the base `Action` class (not specific to `IncludeLaunchDescription`), and is parsed here for frontends: https://github.com/ros2/launch/blob/bdf124d83c847b5205c204932e9f86daccc0c747/launch/launch/action.py#L64
    * Note that the name of the attribute is `if` and not `condition`. `condition` is the name of the corresponding the Python class constructor arg
    * The the
    * You can find an example here if you switch to the "YAML" tabs and search for `if:`: https://docs.ros.org/en/rolling/Tutorials/Intermediate/Launch/Using-Substitutions.html (in this case, there are 2 examples with `executable:`)

I believe it should look like this:

```
- include:
  file: $(find-pkg-share my_package)/launch/optional_components.launch.yaml
  if: $(var use_optional_component)
```

I agree that, if you're not used to the launch system, it's a bit hard to make sense of the source code and figure out how the actual launch file is supposed to look, especially since there aren't a ton of examples.

pitosalas, 2025-03-04:
Thanks. I went with condition: because I found a couple of places where that was in an example ros2.,yaml launch file. Your suggestion, after quoting all the strings, which I think is required, worked. 

```
  - include:
      file: "$(find-pkg-share hls_lfcd_lds_driver)/launch/hlds_laser.launch.py"
      arg:
        - name: port
          value: "/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-port0"
      if: "$(var lidar)"
```
 
I will study the code and come back maybe with a proposal. Thanks

christophebedard, 2025-03-05:
Great! If you can point me to the examples you looked at, I can try to see if they need to be updated/fixed.

pitosalas, 2025-03-06:
Hi Christopher

By the way are you the committer for the launch stuff? And are you the designer? 

I am using the yaml launch file because it is far simpler although the syntax can sometimes be confusing.

The python launch file is powerful but boy oh boy is it involved. It doesn’t seem pythonic more like Java. I don’t know the launch mechanisms nearly enough to propose how to change it but it is an unnecessary steep learning curve really early in one’s experience with ROS2 (although it’s better to have the python format than not like ROS1.)

This is not a rant, just a question about the history of how we got to this particular design.

Pito

christophebedard, 2025-03-12:
Sorry for the delay.

> By the way are you the committer for the launch stuff? And are you the designer?

I'm just one of the people helping maintain launch. Over the years, I've come to learn how it all works.

> I am using the yaml launch file because it is far simpler although the syntax can sometimes be confusing.
>
> The python launch file is powerful but boy oh boy is it involved. It doesn’t seem pythonic [...]

You are definitely not the first person to say this!

I think the main answer is: the Python "interface" was never meant to be the main interface. It's the backend. The YAML and XML launch frontends should be used instead of Python. The Python interface should only be used if someone wants to go deeper _and_ hopefully if they know what they're doing.

Now, this doesn't really help. What's missing is:

1. Point users to YAML and XML launch files in the docs first, _not_ Python
    - Currently, Python seems to be the default option:
        - https://docs.ros.org/en/rolling/How-To-Guides/Launch-file-different-formats.html
            - We may want to revisit the [_Python, XML, or YAML: Which should I use?_ section](https://docs.ros.org/en/rolling/How-To-Guides/Launch-file-different-formats.html#python-xml-or-yaml-which-should-i-use) to lean more towards XML and YAML
        - https://docs.ros.org/en/rolling/Tutorials/Intermediate/Launch/Creating-Launch-Files.html
        - https://docs.ros.org/en/rolling/Tutorials/Intermediate/Launch/Launch-system.html
        - https://docs.ros.org/en/rolling/Tutorials/Intermediate/Launch/Using-Substitutions.html
    - I believe this is because the YAML and XML frontends were not fully implemented when the documentation was first written, but now they should be ready. If they're not, then we can work to improve them.
2. Improve the launch _API_ documentation:
    1. Properly document the names of the actions/substitutions/etc. and their attributes in frontends, since they differ from the corresponding names in the underlying Python classes
        - Users should not have to look at the source code to know that the frontend attribute name of the condition param for an action is `if`
    2. Add examples for all actions/substitutions

Some relevant issues:

1. https://github.com/ros2/launch_ros/issues/404 + https://github.com/ros2/launch_ros/pull/409
    - I had started working on this because of situations similar to yours. I want to pick this back up when I get more time.
    - Once we have this, I want to add more examples to docs.ros.org and link to the relevant `launch`/`launch_ros` docs.
3. https://github.com/ros2/ros2_documentation/issues/4453
5. https://github.com/ros2/ros2_documentation/issues/4188

pitosalas, 2025-03-13:
Interesting. I am surprised I guess that the "backend" is the python api. I thought (maybe this is out of date) that most of ROS2 is in C++ and python is just a patch to allow non-C++ programmers (e.g. like students and early adopters) to use a simpler language.

There are advantages obviously in using Python because everyone knows it and you don't have to reinvent concepts like arguments, conditionals, and also it opens possibilities in terms of customization and flow control. 

But IMO the current Python API is definitely (unnecessarily) hard to use.I wonder if a new one could be built (on top of this one) that simplifies and makes the api and the default launch file more pythonic.

closed 2025-03-13 by christophebedard

pitosalas, 2025-03-13:
Thanks. I am interested in the topic but I don't want to waste your time. 

"And, again, users are meant to use the YAML and XML frontends." -- that may be the case in the develoeprs' minds, but it doesn't come through in the doc, code examples, and functionality available to json and xml launch files. 

I will take a look at the link you sent.  Thanks for engaging on the topic.

christophebedard, 2025-03-14:
> "And, again, users are meant to use the YAML and XML frontends." -- that may be the case in the develoeprs' minds, but it doesn't come through in the doc, code examples, and functionality available [...]

You are definitely right, which is why I'm hoping to change/improve the docs. First step: https://github.com/ros2/ros2_documentation/pull/5093.

> Thanks. I am interested in the topic but I don't want to waste your time.

You're not wasting my time 😁 If you want to help (we're always looking for help!), I think the best way to help here would be to get familiar with the launch system and then help improve the docs like I've outlined above in https://github.com/ros2/launch/issues/832#issuecomment-2718711722. For some actions/substitutions/etc., YAML/XML support might be incomplete, which is part of why there aren't a ton of YAML/XML examples, so improving that would be great too.

pitosalas, 2025-04-04:
As an example, I am trying to do this:"

```
launch:
  - node:
      pkg: nav2_costmap_2d
      exec: costmap_2d_node
      name: local_costmap
      output: screen
      include: "$(find-pkg-share obstacle_perception/config/local_costmap_params.yaml)"
```

How do I do it correctly? And where in the launch source code should I have looked to see if that include is a valid submode of node, or whether I should have a param: key first?

christophebedard, 2025-04-05:
You'd need to understand/find that `include` and `node` are two separate actions, so they're two separate structures. As for arguments of each action (e.g., `node`'s `pkg`/`exec`/etc.), they are parsed with `entity.get_attr()` in the action class' `parse()` function: https://github.com/ros2/launch_ros/blob/d3e2350b29d279ba4b9cbecfdc0db47dfb073e98/launch_ros/launch_ros/actions/node.py#L306 (but you might have to look at the parent class to find the attributes that _it_ parses.

But, again, I know you shouldn't have to know this/look at the source code to figure this out. We're working on improving the docs.

pitosalas, 2025-04-05:
I’m looking at your code in the hopes to be able to contribute. I realized that node and include were different. The sample I included was my last attempt. What I know is that I wanted to give that yaml parameter file to the node. I was guessing how to do that.

christophebedard, 2025-04-05:
Ah! Sorry, I missed that it was about parameters instead of a launch file include.

If you click on the "YAML" tab here and search for `param:` under `node:`, you'll see that you can provide a list of `name`/`value`: https://docs.ros.org/en/rolling/How-To-Guides/Launch-file-different-formats.html#launch-file-examples. `param:` is defined [here](https://github.com/ros2/launch_ros/blob/d3e2350b29d279ba4b9cbecfdc0db47dfb073e98/launch_ros/launch_ros/actions/node.py#L342) in the `Node` action's frontend parsing code.

Also, digging through the source code, I believe you can do this, but I can't find any example in the source code or in the docs:

```yaml
launch:
  - node:
      pkg: nav2_costmap_2d
      exec: costmap_2d_node
      name: local_costmap
      output: screen
      param:
          - from: "$(find-pkg-share obstacle_perception/config/local_costmap_params.yaml)"
```

I believe someone else pointed out that there is no XML/YAML example for this. There would be an example of that in this section _if_ it included a YAML example instead of just a Python example: https://docs.ros.org/en/rolling/Tutorials/Intermediate/Launch/Using-ROS2-Launch-For-Large-Projects.html#using-wildcards-in-yaml-files