ros2/launch #844: Support iterable substitution in PathJoinSubstitution

robin-zealrobotics, 2025-03-23:
This
```python
path = PathJoinSubstitution(
    [FindPackageShare("some_package"), "some_folder", (some_launch_argument, ".more_name.extension")
)
```
Would resolve to
`/path_to_some_package/some_folder/argument.more_name.extension`

Currently, this throws 
`TypeError: Failed to normalize given item of type '<class 'tuple'>', when only 'str' or 'launch.Substitution' were expected.`

I see no obvious reason why this couldn't be supported. As far as I understand, other places that use substitutions support Iterables or Sequence of Substitutions already.

christophebedard, 2025-03-23:
> I see no obvious reason why this couldn't be supported. As far as I understand, other places that use substitutions support Iterables or Sequence of Substitutions already.

Exactly! See https://github.com/ros2/launch/pull/838

christophebedard, 2025-03-23:
I've close this since there's already an issue about this and a PR.

closed 2025-03-23 by christophebedard

robin-zealrobotics, 2025-03-23:
Nice! My apologies, looked for an existing ticket, but didn’t find it.