ros2/launch #849: IncludeLaunchDescription - request name without suffix/extension

emersonknapp, 2025-03-25:
### Description

Allow LaunchDescription inclusion to happen with an incomplete name, having auto-completion fill in the suffix/extension, throwing an error if it's ambiguous.

Examples:

Commandline
```
ros2 launch nav2_bringup bringup
```

Python launchfile
```
from pathlib import Path

from ament_index_python.packages import get_package_share_directory
from launch.actions import  IncludeLaunchDescription

...

  launch_dir = Path(get_package_share_directory('nav2_bringup') / 'launch' 
  ...
  IncludeLaunchDescription(launch_dir / 'bringup'))
```

XML
```
<include file="$(find-pkg-share nav2_bringup)/launch/bringup" />
```

YAML
```
- include:
    file: "$(find-pkg-share nav2_bringup)/launch/bringup"
```

### Motivation

Given that users can write launchfiles in several languages, we have a usability/discoverability problem.

Launchfile authors want to put a syntax-appropriate suffix on the file, so that code editors will highlight the syntax properly.

However from a user perspective, the action of including a launchfile is a semantic one. I am saying "please launch this semantic ros subgraph/system", and I _explicitly_ do not care what language the description of that subsystem was implemented in. Compare: I don't care what language client is publishing the topic I am subscribing to, I should not have to say `subscribe_to_python_publisher('topic_name_py')`.

The discussion of using a single common suffix for launchfiles is pretty split, at least within the core contributor base, so globally using `.launch` is probably never going to be the solution for this.

So, based on some discussions around this limitation, this way of approaching the problem came up.

This allows authors to
1. Properly suffix the file for syntax highlighting in editor
2. Change syntaxes later without breaking users who include the particular launch description

And allows users to
1. Not know or care which language was used to provide a description, focusing on the package's exposed semantic interface

### Design / Implementation Considerations

Probably have to do:
- If file of given name exists, OK
- Else check `path.parent` for files with that prefix
- If there are multiple, throw error
- Else launch that file

Maybe we want to limit this to a specified list of "allowed suffixes" so not any suitable prefix would match?
```
.launch
.xml
.y[a]ml
.py
_launch.[xml|y[a]ml|py]
.launch.[xml|y[a]ml|py]
```

### Additional Information

When implemented, tutorials and demos should be updated to provide this as the default usage pattern.

`ros2launch` CLI should show semantic prefixes as the autocomplete options

emersonknapp, 2025-03-25:
CC @wjwwood @christophebedard @clalancette

christophebedard, 2025-04-23:
As discussed, I think this is good. However, I have a few comments about the strictness of the autocomplete logic. It doesn't really go against the proposal.

>  Probably have to do:
>
> * If file of given name exists, OK
> * Else check path.parent for files with that prefix

Just to make it explicit: if a path with a _full_ filename is given (i.e., with one of the launch file extensions) and that file doesn't exist, I think it should raise an error right away, otherwise the wrong launch file might be loaded. This would probably be the case in practice (filename with typo probably won't match anything else), but I think this should be explicit in the logic.

If the given path doesn't include an extension, then we can look for a file with one of the extensions.

> * If there are multiple, throw error

This part is definitely important.

> Maybe we want to limit this to a specified list of "allowed suffixes" so not any suitable prefix would match?

I think we should do that, yes. We shouldn't try to resolve `dir/my_launch` to `dir/my_launch_very_different.launch.py`.