ros2/launch #856: 我编译ros2_humble 的时候  Could NOT find PythonExtra (missing: PythonExtra_LIBRARIES)

LuFeiX, 2025-04-12:
### Operating System:

```
<PUT OS INFO HERE>
```

### ROS version or commit hash:

humble

### RMW Configuration (if applicable):

Finished <<< rttest [3.00s]
Finished <<< python_qt_binding [9.55s]           
Starting >>> qt_gui_py_common
Starting >>> qt_dotgraph
--- stderr: launch_testing_ament_cmake           
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonExtra (missing: PythonExtra_LIBRARIES)
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /Users/feizhengyang/ros2_humble/install/python_cmake_module/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:224 (find_package_handle_standard_args)
  CMakeLists.txt:29 (find_package)

---
Failed   <<< launch_testing_ament_cmake [6.11s, exited with code 1]
Aborted  <<< qt_dotgraph [0.65s]                            

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

Finished <<< rttest [3.00s]
Finished <<< python_qt_binding [9.55s]           
Starting >>> qt_gui_py_common
Starting >>> qt_dotgraph
--- stderr: launch_testing_ament_cmake           
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonExtra (missing: PythonExtra_LIBRARIES)
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /Users/feizhengyang/ros2_humble/install/python_cmake_module/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:224 (find_package_handle_standard_args)
  CMakeLists.txt:29 (find_package)

---
Failed   <<< launch_testing_ament_cmake [6.11s, exited with code 1]
Aborted  <<< qt_dotgraph [0.65s]                            

### Expected behavior

Finished <<< rttest [3.00s]
Finished <<< python_qt_binding [9.55s]           
Starting >>> qt_gui_py_common
Starting >>> qt_dotgraph
--- stderr: launch_testing_ament_cmake           
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonExtra (missing: PythonExtra_LIBRARIES)
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /Users/feizhengyang/ros2_humble/install/python_cmake_module/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:224 (find_package_handle_standard_args)
  CMakeLists.txt:29 (find_package)

---
Failed   <<< launch_testing_ament_cmake [6.11s, exited with code 1]
Aborted  <<< qt_dotgraph [0.65s]                            

### Actual behavior

Finished <<< rttest [3.00s]
Finished <<< python_qt_binding [9.55s]           
Starting >>> qt_gui_py_common
Starting >>> qt_dotgraph
--- stderr: launch_testing_ament_cmake           
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PythonExtra (missing: PythonExtra_LIBRARIES)
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /Users/feizhengyang/ros2_humble/install/python_cmake_module/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:224 (find_package_handle_standard_args)
  CMakeLists.txt:29 (find_package)

---
Failed   <<< launch_testing_ament_cmake [6.11s, exited with code 1]
Aborted  <<< qt_dotgraph [0.65s]

LuFeiX, 2025-04-12:
PythonExtra 是什么？

MichaelOrlov, 2025-04-24:
@sloretz, could you please comment on this issue?

sloretz, 2025-04-24:
> /Users/feizhengyang/ros2_humble/install/python_cmake_module/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake

> `/Applications/CMake.app/Contents/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake`

It looks like you're attempting to compile ROS Humble on OSX. Please note that ROS Humble on OSX (aka MacOS) is Tier 3 support according to [REP 2000](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027). Tier 3 support means this platform is tested and maintained by members of the community rather than the core ROS team.

Here are some guides and discussions on using ROS Humble with MacOS. They might tell you how to solve this issue:

* https://docs.ros.org/en/humble/Installation/Alternatives/macOS-Development-Setup.html
* https://medium.com/@shubhjain_007/installing-ros2-humble-gazebo-on-mac-m1-m2-m3-silicon-853e0737dcc8
* https://www.reddit.com/r/ROS/comments/1ei5upm/ros2_on_macos_silicon_chips/
* https://github.com/abishek/ros-humble-desktop-m1_2-mac

> PythonExtra 是什么？

PythonExtra is a [CMake Find module](https://cmake.org/cmake/help/latest/manual/cmake-developer.7.html#find-modules) from the [python_cmake_module package](https://github.com/ros2/python_cmake_module/blob/humble/cmake/Modules/FindPythonExtra.cmake). It's mostly used for finding the debug version of the Python interpreter, which is important when building older releases of ROS with debugging symbols on Windows.

A lot has changed in the python_cmake_module package since Humble was released. Someone did try to make python_cmake_module on OSX a long time ago. Maybe there are changes from that old PR that you could try on OSX? https://github.com/ros2/python_cmake_module/pull/1

I will close this for now since this appears to be a Tier 3 support question rather than a bug.

closed 2025-04-24 by sloretz

XQSwx, 2025-09-18:
I am also encountering this issue when using Buildroot for cross-compilation on my computer, which is running Ubuntu 22.04.
`[0.240s] Invoking command in '/root/ros2/build/rclpy': AMENT_PREFIX_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 CMAKE_PREFIX_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 LD_LIBRARY_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/bin:${PATH} PKG_CONFIG_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/x86_64-linux-gnu/pkgconfig:/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/pkgconfig PYTHONPATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/python3.10/site-packages:/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/python3.11/site-packages /usr/bin/cmake /root/ros2/src/ros2/rclpy/rclpy -DCMAKE_PREFIX_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 -DCMAKE_FIND_ROOT_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 -DCMAKE_TOOLCHAIN_FILE=/root/rk3576-linux/buildroot/output/rockchip_rk3576/host/share/buildroot/toolchainfile.cmake -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2
[0.251s] -- Found ament_cmake: 1.3.12 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/ament_cmake/cmake)
[0.352s] -- Found lifecycle_msgs: 1.2.2 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/lifecycle_msgs/cmake)
[0.368s] -- Found rosidl_generator_c: 3.1.7 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rosidl_generator_c/cmake)
[0.370s] -- Found rosidl_adapter: 3.1.7 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rosidl_adapter/cmake)
[0.375s] -- Found rosidl_generator_cpp: 3.1.7 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rosidl_generator_cpp/cmake)
[0.383s] -- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
[0.394s] -- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
[0.405s] -- Found rcl: 5.3.10 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rcl/cmake)
[0.430s] -- Found rmw_implementation_cmake: 6.1.2 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rmw_implementation_cmake/cmake)
[0.431s] -- Found rmw_fastrtps_cpp: 6.2.8 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rmw_fastrtps_cpp/cmake)
[0.485s] -- Using RMW implementation 'rmw_fastrtps_cpp' as default
[0.491s] -- Found rcl_action: 5.3.10 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rcl_action/cmake)
[0.514s] -- Found rcl_lifecycle: 5.3.10 (/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/share/rcl_lifecycle/cmake)
[0.556s] CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
[0.556s]   Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Development
[0.556s]   Development.Module Development.Embed) (found suitable version "3.11.8",
[0.556s]   minimum required is "3.11")
[0.556s] Call Stack (most recent call first):
[0.556s]   /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
[0.557s]   /usr/share/cmake-3.22/Modules/FindPython/Support.cmake:3180 (find_package_handle_standard_args)
[0.557s]   /usr/share/cmake-3.22/Modules/FindPython3.cmake:490 (include)
[0.557s]   CMakeLists.txt:41 (find_package)
[0.557s] 
[0.557s] 
[0.557s] -- Configuring incomplete, errors occurred!
[0.557s] See also "/root/ros2/build/rclpy/CMakeFiles/CMakeOutput.log".
[0.564s] Invoked command in '/root/ros2/build/rclpy' returned '1': AMENT_PREFIX_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 CMAKE_PREFIX_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 LD_LIBRARY_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/bin:${PATH} PKG_CONFIG_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/x86_64-linux-gnu/pkgconfig:/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/pkgconfig PYTHONPATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/python3.10/site-packages:/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2/lib/python3.11/site-packages /usr/bin/cmake /root/ros2/src/ros2/rclpy/rclpy -DCMAKE_PREFIX_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 -DCMAKE_FIND_ROOT_PATH=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2 -DCMAKE_TOOLCHAIN_FILE=/root/rk3576-linux/buildroot/output/rockchip_rk3576/host/share/buildroot/toolchainfile.cmake -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/root/rk3576-linux/buildroot/output/rockchip_rk3576/target/opt/ros2
`